GhostESP-Revival/GhostESP #5: ESP32 C3 RGB LED not lighting up

thelouisgan, 2025-05-04:
Flashed latest 1.4.9 firmware on my ESP32-C3-WROOM-02 and all is working well, however the built in RGB LED on pin 8 doesn't seem to light up at all with any mode (rainbow, strobe etc) I tried setting the rgb pins to 8 8 8 ```> setrgbpins 8 8 8
RGBManager deinitialized (separate pins)
RGBManager initialized for pin 8 with 1 LEDs
Single-pin RGB configured on GPIO 8!``` but when I reboot it still shows `RGBManager initialized for separate R/G/B pins: 0, 0, 0`

The esp32 is connected to my flipper zero using the Ghost ESP app, and the built-in app LED configs also do not seem to turn the light on.

Light is working as I tested with a previous program so the LED is not fried

Thank you for reviving GhostESP

Boot log:
```
> setrgbpins 8 8 8
RGBManager deinitialized (separate pins)
RGBManager initialized for pin 8 with 1 LEDs
Single-pin RGB configured on GPIO 8!
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x145c
load:0x403cc710,len:0xb9c
load:0x403ce710,len:0x2e4c
entry 0x403cc71a
I (30) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (30) boot: compile time Oct 26 2024 12:25:04
I (30) boot: chip revision: v0.4
I (31) boot.esp32c3: SPI Speed      : 80MHz
I (35) boot.esp32c3: SPI Mode       : DIO
I (38) boot.esp32c3: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00007000
I (62) boot:  1 app0             factory app      00 00 00010000 003d0000
I (69) boot:  2 coredump         Unknown data     01 03 003e0000 00020000
I (75) boot: End of partition table
I (78) esp_image: segment 0: paddr=00010020 vaddr=3c0f0020 size=56ad8h (355032) map
I (141) esp_image: segment 1: paddr=00066b00 vaddr=3fc96800 size=03c78h ( 15480) load
I (144) esp_image: segment 2: paddr=0006a780 vaddr=40380000 size=05898h ( 22680) load
I (149) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=ebd58h (965976) map
I (304) esp_image: segment 4: paddr=0015bd80 vaddr=40385898 size=10ef0h ( 69360) load
I (324) boot: Loaded app from partition at offset 0x10000
I (324) boot: Disabling RNG early entropy source...
Serial Started...
[0;33mW (536) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration[0m
[0;33mW (570) phy_init: saving new calibration data because of checksum failure, mode(2)[0m
WiFi started.
Ready to scan.
Global CA certificate store initialized successfully.
Registered Commands
[0;33mW (585) SETTINGS: Using default RGB mode[0m
Failed to load Channel Delay: ESP_ERR_NVS_NOT_FOUND
Failed to load AP SSID
Failed to load AP Password
Failed to load Portal URL
Failed to load Portal SSID
Failed to load Portal Password
Failed to load Portal AP SSID
Failed to load Portal Domain
Failed to load Printer IP
Failed to load Printer Text
Failed to load Flappy Ghost Name
Failed to load Hex Accent Color String
Settings loaded from NVS.
Settings loaded successfully.
Wi-Fi already initialized, skipping Wi-Fi init.
DHCP server configured successfully.
Wi-Fi Access Point started with SSID: GhostNet
mDNS hostname set to ghostesp.local
HTTP server started
ESP32 AP IP Address: 
192.168.4.1NVS namespace 'sd_config' not found. Using default SD pins.
SD Card Pin Configuration:
MMC Mode:
  CLK: GPIO19
  CMD: GPIO18
  D0:  GPIO20
  D1:  GPIO21
  D2:  GPIO22
  D3:  GPIO23
SPI Mode:
  CS:   GPIO0
  CLK:  GPIO0
  MISO: GPIO0
  MOSI: GPIO0
SD Card init failed with loaded pins. Check configuration.
RGBManager initialized for separate R/G/B pins: 0, 0, 0
Ghost ESP Ready ;)
```

jaylikesbunda, 2025-05-06:
i've changed how the rgb pins are saved and loaded in the [latest pre-release](https://github.com/jaylikesbunda/Ghost_ESP/releases/tag/prerelease-20250505-163338-8867e16), please try it out and let me know how it goes

thelouisgan, 2025-05-06:
[image]

[image]

[image]

It is working perfectly now; many thanks!

73

closed 2025-05-06 by thelouisgan