GhostESP-Revival/GhostESP #68: CYD Issue: Access Point list fails to list correctly

ProjectNemoSan, 2025-07-11:
The issue persists on all 5 variants of my displays regardless of install method or if I do a compile from source. I go to scan, it scans and stops at mid 30-65. When I scroll up the the list fails to recall the first portion of the list and if it does list it skips multiple aps or or nothing at all. If screen is blank it usually fails to list anywhere from 20-30 aps.

jaylikesbunda, 2025-07-12:
Would you prefer it to cut off the last aps when it runs out of space? I can see what i can do about giving it more memory but we are fairly constrained already

tototo31, 2025-07-12:
Might make more sense to have it list from the top down honestly - Strongest APs first then if we need to ditch any cause of mem constraints its the ones with the weaker signals that get chopped vs the stronger ones

ProjectNemoSan, 2025-07-13:
Id say either by RSSI or even by channel.  But Im also cool with it not trying to scan 60 aps either.